WatershedArts/Footfall #107: /home/dev05/of_v0.9.8_linux64_release/addons/ofxHttpUtils/src/ofxHttpTypes.h:51:43: error: invalid initialization of reference of type ‘const string& {aka const std::__cxx11::basic_string<char>&}’ from expression of type ‘const boost::filesystem::path’    formFiles[fieldName] = ofToDataPath(path);

SourabhMane, 2018-01-29:
Please help me in resolving this  problem, i m stuck on this compiling error for many days.

Anand02, 2018-01-31:
ok exactly what is the probelem/error ?

vssystemluba, 2018-04-16:
Same issue here.  openFrameworks 0.9.8, gcc/g++ version 4.9.2 on Jessie (Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux).  Anyone seen this before or know of a possible workaround?

Thanks!

vssystemluba, 2018-04-16:
Okay, I was able to at least compile the app by changing the following line in `~/openFrameworks/addons/ofxHttpUtils/src/ofxHttpTypes.h`:
from
```
formFiles[fieldName] = ofToDataPath(path);
```
to
```
formFiles[fieldName] = ofToDataPath(path.string());
```

I assume that this will break something down the line, but at least Footfall is compiling for me now.  :)

closed 2018-08-23 by DHaylock