WatershedArts/Footfall #109: Raspbian compilation issues with osxCsv

vssystemluba, 2018-04-11:
Hello,

I'm having trouble compiling on Raspbian (on an RPi 3).  I followed the openFrameworks installation guide at https://forum.openframeworks.cc/t/compiling-of-in-raspbian-stretch/27562/15 successfully, but when I try and make Footfall (which names ofxCsv as a dependency) the make fails with the following message:

```
...
In file included from /home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.h:33:0,
                          from /home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:31:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:19: error: 'string' does not name a type
  ofxCsvRow(const string &cols, const string &separator=",");
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:39: error: 'string' does not name a type
  ofxCsvRow(const string &cols, const string &separator=",");
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:48:19: error: 'vector' does not name a type
  ofxCsvRow(const vector<string> &cols);
...
...
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsvRow.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsvRow.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
```

I'm using oF version 0.9.8 and gcc 6.3 on Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux.

Any help or information you can give is greatly appreciated.

Thanks!

vssystemluba, 2018-04-16:
I've since switched off of raspbian stretch and am now compiling with raspbian jessie successfully. I recommend biting the bullet and using the slightly older OS if Footfall is the only application you're running on your pi.

closed 2018-04-16 by vssystemluba

Chesterguan, 2018-10-11:
Hi There  I am trying to repeat the Footfall project right now, and i got the same problem,could u please write a detailed guide? I have a Raspberry Pi 3 and dont know which OS to install and which install guides to follow

vssystemluba, 2018-10-12:
@Chesterguan I followed the exact documentation linked from the README.md of this repository.  In the first step where they say to download an OS with NOOBS, just make sure you wind up installing Raspbian Jessie and NOT the newer Raspbian Stretch.  The rest of the documentation in this repo is pretty complete and I was able to get up and running after installing Raspbian Jessie.

Chesterguan, 2018-10-12:
Gotcha , thanks so much for your time!!

Chesterguan, 2018-10-14:
oh lord i met the new problem that I have installed the Jessie 06.12 as OS and I follow the  link below to compile the openFrameworks: https://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/
Everything seems great till I tried to compile the Footfall project. 
The Problem is "ofxVideoBaseTypes.h no such file or directory". It seems the ofxCv folder is not complete. And search for a long time, there is no solution till now. Have u met the problem, and how could u solved it ? Thanks so much. It is my first time to play with rpi and felt so hurt by facing so many problems. 

Thanks for your time !! best regards