WatershedArts/Footfall #110: Compatibility with Stretch

vertigoRod, 2018-05-15:
Hi,
Did you guys have a chance to run it with RASPBIAN STRETCH?

naraquev, 2018-06-02:
Yes! I was able to run it with Stretch

c1audiu, 2018-06-11:
naraquev, can you help us with some info regarding the instalation on stretch? Like what version of openframeworks did you installed and other particularities. I olny managed to make it run on raspberrypi zero w with 2017-07-05-raspbian-jessie-lite and of_v0.9.8_linuxarmv6l_release.tar.gz. Thank you in advanced.

black23, 2018-07-19:
@naraquev Please Can you say to us how to compile Footfall on Stretch?

naraquev, 2018-07-19:
Hello all!
I was able to run it on Stretch with OP 0.10. There are several steps to do so. The CsvManager and HTTPManager needs several modifications. In my case I simply deleted the HTTPManager because I wanted to communicate with MQTT, with the CsvManager you have to check the using std and namespaces of the String types.

I hope this help you all!

Ferni492, 2018-08-06:
@naraquev Please can you share the rpi3 image???

DHaylock, 2018-08-23:
It is compatible with Raspbian Stretch you need to use openFrameworks 0.10. There are a number of caveats, You need to include the ofxPoco addon as without it ofxHttpUtils breaks. Everything else should work fine.

closed 2018-08-23 by DHaylock

mobileonekh, 2018-10-12:
OpenFrameworks is a main blocking point.

@DHaylock and @naraquev  Can you write more detail or give some link on how to get OpenFramework up and running on Stretch?

Thanks