WatershedArts/Footfall #113: Raspberri Pi 3 Compiling Error

Ferni492, 2018-08-07:
Hi, I have this error when compiling
In file included from /home/pi/openFrameworks/addons/ofxCv/src/ofxCv.h:9:0,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingManager.h:12,
                 from /home/pi/openFrameworks/apps/Footfall/Footfall/src/TrackingManager.cpp:7:
/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/include/ofxCv/Utilities.h:16:30: fatal error: ofVideoBaseTypes.h: No such file or directory
 #include "ofVideoBaseTypes.h"

Ferni492, 2018-08-08:
The solution was the new stable branch of ofxCv, then I had some errors with ofxHttpUtils which I solved with the "-gh-pages" version. Good luck to everyone.

closed 2018-08-08 by Ferni492

Chesterguan, 2018-10-14:
Hi I got the same problem and where can I get the stable branch of ofxCv, you mean the link below ?
https://github.com/kylemcdonald/ofxCv

Thanks for your time