WatershedArts/Footfall #114: Threshold

gomezramones, 2018-08-23:
Hi, I have a doubt on your code because you use two threshold, the first for MOG2 function and the second is for the processed Mog image and that is my question. Why did you use the second threshold for the processed MOG image? Because that image is a binary image, isn't it?

DHaylock, 2018-08-23:
Hi @gomezramones,

The [second threshold](https://github.com/WatershedArts/Footfall/blob/master/Footfall/src/CameraManager.cpp#L125) is used because we blur the image after its been processed.

David

closed 2018-08-23 by DHaylock

gomezramones, 2018-08-23:
Ok DHaylock, but that image is all ready on binary, right? So why use a threshold when the image is 0 or 255

oliverhumpage, 2018-08-23:
> On 23 Aug 2018, at 15:32, gomezramones <notifications@github.com> wrote:
> 
> Ok DHaylock, but that image is all ready on binary, right? So why use a threshold when the image is 0 or 255

The blur will turn the image back into greyscale, so it is no longer monochrome at that point. Blurring then re-thresholding is a valid operation.

While the image processing code may look a bit convoluted, it has been created through experimentation in our specific setup. You may find that tweaking it produces a better result for your situation.

gomezramones, 2018-08-23:
Thanks for the answers oliverhumpage and DHaylock. 

Now, I'm testing the code with a camera and I have changed some values to get better results, however the only problem that I got are the shadows since the room has different light (Sun, light bulb) causing some shadows and the algorithm thinks that are a persons.

Thank you for the help.

My best regards

gomezramones, 2018-08-23:
Forgot to explain the above question that I made. You have some parameter to change but there aren't to much to change when it comes to the light. Only there is usemask, minradius and other aspectos but none of them change the effect of the light, so what can i do?

DHaylock, 2018-08-24:
Hi @gomezramones,

Unfortunately, you will need to do some tweaks in the parameters. And it is trial and error.
The shadow issue should be solved with the [track shadows and shadow pixel ratio options](https://github.com/WatershedArts/Footfall/blob/master/docs/configuration.md)

Otherwise I would increase the moghistory. 
Failing that it a matter of shifting your camera.

David

gomezramones, 2018-08-27:
Thanks for the reply Mr. DHaylock, it was very helpfull. Sorry for ask many questions but If I want to do some pre-processing before tracking object (clahe, changing histogram, correction gamma). Which is the variable that was assigned to the camara? Or more specifically where do you open webcam?

My best regards

gomezramones, 2018-08-27:
Can I change a little bit the code that you made it? Because I found videoPlayer is the variable that I have to modify but when I run the program it doesn't change anything