WatershedArts/Footfall #115: 'HTTPManager.o' failed

Ferni492, 2018-08-23:
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linuxarmv6l/Release/src/HTTPManager.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/HTTPManager.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2

DHaylock, 2018-08-24:
Hi @Ferni492,

Can you copy more of the error log. It's hard to tell why its failing from this log.

David

rlnunez, 2018-10-24:
I am having similar issues, here is the output when the errors start

`/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:190:71: warning: ‘bestMatchGtVal’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                 if(curFocalLength > focalLength && (bestMatchGt == -1 || curFocalLength < bestMatchGtVal)){
                                                    ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/openFrameworks/addons/ofxCv/libs/ofxCv/src/Calibration.cpp:185:73: warning: ‘bestMatchLtVal’ may be used uninitialized in this function [-Wmaybe-uninitialized]
                 if(curFocalLength <= focalLength  && (bestMatchLt == -1 || curFocalLength > bestMatchLtVal)){
                                                      ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:31:0:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:19: error: ‘string’ does not name a type
   ofxCsvRow(const string &cols, const string &separator=",");
                   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:39: error: ‘string’ does not name a type
   ofxCsvRow(const string &cols, const string &separator=",");
                                       ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:48:19: error: ‘vector’ does not name a type
   ofxCsvRow(const vector<string> &cols);
                   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:48:25: error: expected ‘,’ or ‘...’ before ‘<’ token
   ofxCsvRow(const vector<string> &cols);
                         ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:64:19: error: ‘string’ does not name a type
   void load(const string &cols, const string &separator=",");
                   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:64:39: error: ‘string’ does not name a type
   void load(const string &cols, const string &separator=",");
                                       ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:71:19: error: ‘vector’ does not name a type
   void load(const vector<string> &cols);
                   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:71:25: error: expected ‘,’ or ‘...’ before ‘<’ token
   void load(const vector<string> &cols);
                         ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:88:3: error: ‘string’ does not name a type
   string toString(bool quote=false, const string &separator=",");
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:91:10: error: ‘ostream’ does not name a type
   friend ostream& operator<<(ostream &ostr, const ofxCsvRow &row);
          ^~~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:116:3: error: ‘string’ does not name a type
   string getString(int col) const;
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:139:18: error: ‘string’ has not been declared
   void addString(string what);
                  ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:173:27: error: ‘string’ has not been declared
   void setString(int col, string what);
                           ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:208:30: error: ‘string’ has not been declared
   void insertString(int col, string what);
                              ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:228:3: error: ‘vector’ does not name a type
   vector<string>& getData();
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:237:3: error: ‘vector’ does not name a type
   vector<string>::iterator begin();
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:238:3: error: ‘vector’ does not name a type
   vector<string>::iterator end();
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:239:3: error: ‘vector’ does not name a type
   vector<string>::const_iterator begin() const;
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:240:3: error: ‘vector’ does not name a type
   vector<string>::const_iterator end() const;
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:241:3: error: ‘vector’ does not name a type
   vector<string>::reverse_iterator rbegin();
   ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:242:3: error: ‘vector’ does not name a type
[... 288 lines cut ...]
              ^~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:317:1: error: expected primary-expression before ‘}’ token
 }
 ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:317:1: error: expected  ;’ before ‘}’ token
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:317:1: error: expected primary-expression before ‘}’ token
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:317:1: error: expected  )’ before ‘}’ token
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:317:1: error: expected primary-expression before ‘}’ token
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp: At global scope:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:320:1: error: ‘string’ does not name a type
 string trimString(const string &s) {
 ^~~~~~
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.cpp:335:1: error: ‘vector’ does not name a type
 vector<string> ofxCsvRow::fromString(const string &row, const string &separator) {
 ^~~~~~
[... cut ...]
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsvRow.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsvRow.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
`