WatershedArts/Footfall #116: compiling error

gomezramones, 2018-08-28:
I have changed some part in the code on CameraManager.cpp, the void CameraManager::update() function. I put

In file included from /home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCv/src/ofxCv.h:23:0,
                 from src/CameraManager.h:17,
                 from src/CameraManager.cpp:7:
/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Tracker.h: In member function ‘virtual const std::vector<unsigned int>& ofxCv::RectTracker::track(const std::vector<cv::Rect_<int> >&)’:
/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCv/libs/ofxCv/include/ofxCv/Tracker.h:326:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    for(int i = 0; i < labels.size(); i++) {
                   ~~^~~~~~~~~~~~~~~
src/CameraManager.cpp: In member function ‘void CameraManager::setup(Camera_Configuration)’:
src/CameraManager.cpp:18:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < _cameraConfig.maskCoord.size(); i++)
                   ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/CameraManager.h:20:0,
                 from src/CameraManager.cpp:7:
/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCvPiCam/src/ofxCvPiCam.h: At global scope:
/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCvPiCam/src/ofxCvPiCam.h:200:13: warning: ‘void color_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void color_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^~~~~~~~~~~~~~
/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/addons/ofxCvPiCam/src/ofxCvPiCam.h:201:13: warning: ‘void gray_callback(MMAL_PORT_T*, MMAL_BUFFER_HEADER_T*)’ declared ‘static’ but never defined [-Wunused-function]
 static void gray_callback(MMAL_PORT_T *, MMAL_BUFFER_HEADER_T *);
             ^~~~~~~~~~~~~
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:221: recipe for target 'obj/linuxarmv6l/Release/src/CameraManager.o' failed
make[1]: *** [obj/linuxarmv6l/Release/src/CameraManager.o] Error 2
make[1]: Leaving directory '/home/pi/openFrameworks/of_v0.10.0_linuxarmv6l_release/apps/Footfall/Footfall'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed

luiscesjr, 2019-01-23:
I am running into the same error, but somewhat different circumstances. I simply cloned git and executed make as per instruction:

```
pi@raspberrypi:~/openFrameworks/apps/Footfall/Footfall $ make -j3
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0
with PKG_CONFIG_LIBDIR=
Compiling OF library for Release
make[1]: Entering directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0
with PKG_CONFIG_LIBDIR=
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0
with PKG_CONFIG_LIBDIR=
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0
with PKG_CONFIG_LIBDIR=
Done!
make[1]: Leaving directory '/home/pi/openFrameworks/libs/openFrameworksCompiled/project'

Compiling Footfall for Release
make[1]: Entering directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal libcurl glfw3 rtaudio gtk+-3.0
with PKG_CONFIG_LIBDIR=
make[1]: *** No rule to make target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/./src/CsvManager.o', needed by 'bin/Footfall'.  Pare.
make[1]: ** Esperando que outros processos terminem.
cat: obj/linuxarmv6l/Release/.compiler_flags: Arquivo ou diretório não encontrado
make[1]: Leaving directory '/home/pi/openFrameworks/apps/Footfall/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2

```

iverdugos, 2019-06-24:
https://github.com/texane/stlink/issues/625

first 'make', then, after the error,  'make clean'. Finally 'make' again. It will compile. At least it did on my Rasberry Pi 3 model B