WatershedArts/Footfall #119: Source code error within ofxCsv.cpp

sorlax2410, 2018-11-08:
[errors.txt](https://github.com/WatershedArts/Footfall/files/2560725/errors.txt)

sorlax2410, 2018-11-08:
I am using OpenFramework v0.10.0_linuxarmv6l_release. RPI version 4.9, armv6L

sorlax2410, 2018-11-08:
The error of version 0.9.8 is cannot find ofVideoBaseTypes.h in ofxCv

sorlax2410, 2018-11-08:
During the use of OpenFrameworks v0.9.8, the same errors occurred just like issue below:
https://github.com/kylemcdonald/ofxFaceTracker-iOS/issues/1

Now trying OF nightly build as the suggested solution

sorlax2410, 2018-11-09:
Met the same error as the v0.10.0

prisikarm, 2019-02-21:
having the same ofxCsv.cpp compilation error (PI3B, of0.10.0, jessie)

`In file included from /home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.h:33:0,
                 from /home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:31:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:19: error: ‘string’ does not name a type
   ofxCsvRow(const string &cols, const string &separator=",");
                   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:45:39: error: ‘string’ does not name a type
   ofxCsvRow(const string &cols, const string &separator=",");
                                       ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:48:19: error: ‘vector’ does not name a type
   ofxCsvRow(const vector<string> &cols);
                   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:48:25: error: expected ‘,’ or ‘...’ before ‘<’ token
   ofxCsvRow(const vector<string> &cols);
                         ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:64:19: error: ‘string’ does not name a type
   void load(const string &cols, const string &separator=",");
                   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:64:39: error: ‘string’ does not name a type
   void load(const string &cols, const string &separator=",");
                                       ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:71:19: error: ‘vector’ does not name a type
   void load(const vector<string> &cols);
                   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:71:25: error: expected ‘,’ or ‘...’ before ‘<’ token
   void load(const vector<string> &cols);
                         ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:88:3: error: ‘string’ does not name a type
   string toString(bool quote=false, const string &separator=",");
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:91:10: error: ‘ostream’ does not name a type
   friend ostream& operator<<(ostream &ostr, const ofxCsvRow &row);
          ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:116:3: error: ‘string’ does not name a type
   string getString(int col) const;
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:139:18: error: ‘string’ has not been declared
   void addString(string what);
                  ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:173:27: error: ‘string’ has not been declared
   void setString(int col, string what);
                           ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:208:30: error: ‘string’ has not been declared
   void insertString(int col, string what);
                              ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:228:3: error: ‘vector’ does not name a type
   vector<string>& getData();
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:237:3: error: ‘vector’ does not name a type
   vector<string>::iterator begin();
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:238:3: error: ‘vector’ does not name a type
   vector<string>::iterator end();
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:239:3: error: ‘vector’ does not name a type
   vector<string>::const_iterator begin() const;
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:240:3: error: ‘vector’ does not name a type
   vector<string>::const_iterator end() const;
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:241:3: error: ‘vector’ does not name a type
   vector<string>::reverse_iterator rbegin();
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:242:3: error: ‘vector’ does not name a type
   vector<string>::reverse_iterator rend();
   ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsvRow.h:243:3: error: ‘vector’ does not name a type
   vector<string>::const_reverse_iterator rbegin() const;
   ^
[... 504 lines cut ...]
 }
 ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp: In member function ‘ofxCsvRow& ofxCsv::front()’:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:364:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp: In member function ‘ofxCsvRow& ofxCsv::back()’:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:369:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp: In member function ‘size_t ofxCsv::size() const’:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:374:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp: In member function ‘bool ofxCsv::empty() const’:
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:379:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus: some warnings being treated as errors
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: recipe for target '/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsv.o' failed
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsv.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks/apps/myApps/Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
`

bgmario, 2019-03-05:
You need to add this line :
using namespace std; 
At the beginning of each file in the folder "/openFrameworks/addons/ofxCsv/src"
And then you also have to change the function "MAX" to "max" in lowercase in every file that uses that function.

kushaldalsania, 2019-03-13:
cc1: warning: command line option ‘-std=c++1y’ is valid for C++/ObjC++ but not for C
cc1: warning: command line option ‘-std=c++1y’ is valid for C++/ObjC++ but not for C
In file included from /home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.c:32:0:
/home/pi/openFrameworks/addons/ofxCvPiCam/libs/mmal/src/interface/mmal/core/mmal_port_private.h:93:67: error: unknown type name ‘MMAL_CLOCK_PAYLOAD_T’
 typedef void (*MMAL_PORT_CLOCK_EVENT_CB)(MMAL_PORT_T *port, const MMAL_CLOCK_PAYLOAD_T *event);
                                                                   ^
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCvPiCam/libs/mmal/src/interface/mmal/util/mmal_graph.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/home/pi/openFrameworks/addons/ofxHttpUtils/src/ofxHttpUtils.cpp: In constructor ‘ofxHttpUtils::ofxHttpUtils()’:
/home/pi/openFrameworks/addons/ofxHttpUtils/src/ofxHttpUtils.cpp:50:87: error: no matching function for call to ‘Poco::Net::Context::Context(Poco::Net::Context::Usage, const char [1], Poco::Net::Context::VerificationMode)’
      Context::Ptr pContext = new Context(Context::CLIENT_USE, "", Context::VERIFY_NONE);
                                                                                       ^
/home/pi/openFrameworks/addons/ofxHttpUtils/src/ofxHttpUtils.cpp:50:87: note: candidate is:
In file included from /usr/include/Poco/Net/SSLManager.h:45:0,
                 from /home/pi/openFrameworks/addons/ofxHttpUtils/src/ofxHttpUtils.cpp:12:
/usr/include/Poco/Net/Context.h:106:2: note: Poco::Net::Context::Context(Poco::Net::Context::Usage, const string&, const string&, const string&, Poco::Net::Context::VerificationMode, int, bool, const string&)
  Context(
  ^
/usr/include/Poco/Net/Context.h:106:2: note:   candidate expects 8 arguments, 3 provided
make[1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxHttpUtils/src/ofxHttpUtils.o] Error 1
make: *** [Release] Error 2

bgmario, 2019-03-13:
You need ofxPoco addon
Try this:
Edit  the file addons.make (openFrameworks/apps/Footfall/Footfall/addons.make) by adding this line before ofxHttpUtils: ofxPoco

kushaldalsania, 2019-03-14:
> You need ofxPoco addon
> Try this:
> Edit the file addons.make (openFrameworks/apps/Footfall/Footfall/addons.make) by adding this line before ofxHttpUtils: ofxPoco

Still same compile errors with your above suggested fix. I see the issue is with version of Poco. 
Same compilation error also occurs in 'develop' branch.
Please send me the hardware and software information on which this footfall repo compiles fine. So that I can replicate same over here.

Below is software and hardware information that I am using for footfall:
1] RPi3 B with Raspbian jessie (tried with Raspbian Stretch too)
2] PiCam v2
3] openFramework v0.10.0 (**arm6** version)
4] All other addons are as per latest release.

bgmario, 2019-03-14:
The hardware and software I use is:
1) RPi3 B+ with  Raspbian Stretch
2) PiCam v2
3) openFramework v0.10.1 (arm6 version)

The mistake you have, I had it  and is because it does not compile ofxHttpUtils in version 0.10.X of OF and therefore you need to use ofxPoco, what I see is that you use Jessie and the recommended version of OF is 0.9.8 (with which it is not necessary to have ofxPoco).
Have you tried OF version 0.9.8?