WatershedArts/Footfall #120: Copiling rpi3 of_0.10.0

collote69, 2019-01-31:
Hi, I wanted to know if someone has solved the Compiling error.
/home/pi/openFrameworks/addons/ofxCsv/src/ofxCsv.cpp:427:2: error: 'data' was not declared in this scope
   data [row] .expand (cols);
   ^ ~~~
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: Failed instructions for the target '/ home / pi / openFrameworks / addons / obj / linuxarmv6l / Release / ofxCsv / src / ofxCsv.o '
make [1]: *** [/home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxCsv/src/ofxCsv.o] Error 1
make [1]: it leaves the directory '/ home / pi / openFrameworks / apps / Footfall / Footfall'
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: Failed instructions for the 'Release' objective
make: *** [Release] Error 2
Thank you

collote69, 2019-02-15:
...

prisikarm, 2019-02-21:
struggling with the same issue, did you managed this error already?