WatershedArts/Footfall #122: Compiling at Rpi3 need ages/stuck at main.cpp

LaurenzGebauer, 2019-03-20:
Hello, first thank you for your work.

I was able to install the newest version of OF on my raspberry 3 using the newest rasbian noobs version 3.0. Tried the camera test, working. 
On the compile step with the command make - j3 it stucks at main.cpp I waited for about 3 hours now but it still hasn't changed yet. No errors but at this step my raspberry is super slow so that i can barley move the mouse. Since this Is possible I though maybe it needs its time to compile 

[image]

but unfortunately after 3 hours I guess there must be a problem somewhere. Anybody have an idea?

Best regards

lancelon, 2020-11-07:
Did you ever get this working?

laurenzraccon, 2020-11-08:
No I didnt, i ended up using an other project.

lancelon, 2020-11-08:
Hiya - would you share what you used instead? Would you recommend it? Thank you. Wasn't Camlytics was it?

zaanen, 2020-11-11:
hmm i can compile on a rasp 4 but when all things set up and i just use startup script it all stays black. Would love to knuw what other script you are now using